PHPirates/SolArduino #46: Add option to plan storm-mode ahead

PHPirates, 2016-12-24:


slideclimb, 2016-12-30:
- [ ] remember if panels were in auto mode 
    - [ ] if auto mode: switch back to auto mode afterwards (at end time)
    - [ ] if not auto mode: set panels at angle it was before
        - [ ] remember angle
- [ ] remember begin and start time
- [ ] set panel to 0 at start time

slideclimb, 2016-12-31:
**Android**

- [ ] convert date/time to time in **seconds** since 1970
- [ ] send request ```/?stormstart=----------&stormend=----------```

slideclimb, 2016-12-31:
**Desktop**

- [ ] convert date/time to time in **seconds** since 1970
- [ ] send request ```/?stormstart=----------&stormend=----------```
    - [ ] times in url are ```UnixTime[] + 3600``` when calculating with MMa

PHPirates, 2018-07-06:
Moved to a note in the GitHub Project because of too low priority.

closed 2018-07-06 by PHPirates